STMicroelectronics/STM32CubeC0 #2: ST Link driver installation issue

A23Velvizhi, 2024-02-28:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * The board (STM32 - NUCLEOC031C6)
 * IDE - keiluvision Version - 5
 
**Describe the bug**
Unable to install the ST link driver 

**Screenshots**
[image]

A23Velvizhi, 2024-02-28:
Procedure:
   In device manager >other devices>ST-Link Debug>Update drive

[image]
[image]

A23Velvizhi, 2024-02-28:
[image]

kindly check

ALABSTM, 2024-03-01:
Hi @A23Velvizhi,

As mentioned in the **CONTRIBUTING.md** guide, this support channel is exclusively dedicated to issues related to the software published within this repository.

https://github.com/STMicroelectronics/STM32CubeC0/blob/e23a3ad670a74f2b6f9257801e84262046909071/CONTRIBUTING.md?plain=1#L38-L40

I suggest to submit your question to the [ST Community](https://community.st.com/s/topiccatalog). There, you will surely find the help you need to overcome your problem.

Please allow me to close this thread. Thank you for your comprehension and good luck.

With regards,

closed 2024-03-01 by ALABSTM