STMicroelectronics/STM32CubeC0 #5: I2C Master Memory Read function actually sends a write command

AndrewWern, 2025-08-29:
**Hardware**
Exact processor: STM32C092CCU6 on a custom PCB
Interfacing with TLE493DW2B6A1HTSA1 hall magnetic vector sensor

**Describe the bug**
section 2.1.3.1 of the Infineon user manual for this sensor shows a timing diagram in Figure 7. The master is required to send the 7-bit device address, followed by a read bit (value 1) for the first byte, then 8 more control bits including the register address to be read. Function HAL_I2C_Mem_Read_IT in stm32c0xx_hal_i2c.c line 2945 calls I2C_TransferConfig with the final parameter I2C_GENERATE_START_WRITE which causes the 7-bit address to be followed by a write bit (value 0) instead.

It is my understanding, the purpose of HAL_I2C_Mem_Read_IT  is to allow the processor to control the SDA line for 2 or 3 bytes before releasing control for the slave to take over, which makes it ideal for the 2-byte read mode in this sensor. I have changed I2C_GENERATE_START_WRITE as referenced above to I2C_GENERATE_START_READ, and the bug goes away. I was further able to see that register bit I2Cx->CR2->RD_WRN was not set before my fix, and was set after my fix.

KRASTM, 2025-09-03:
Hello @AndrewWern,

Thank you for the report.

Regarding the issue, and after a first analysis, the API HAL_I2C_Mem_Read_IT() is used to read data from a specific memory address. The first step after the start condition is to write the memory address (send the slave address with the write bit (value 0 `as you mentioned`), then configure the memory address size usually 1 or2 bytes. The 4th parameter indicates the Reload End mode used which is I2C_SOFTEND_MODE "this prevent automatic STOP generation after sending the memory address and allows the driver to generate a repeated START later for the read phase". This is why we have the I2C_GENERATE_START_WRITE request in the I2C_TransferConfig() function, after all these steps, the slave responds by sending the requested data bytes. The Read phase it happens inside the I2C_Mem_ISR_IT() interrupt subroutine which is configured on line 2923.

For the bit `I2Cx->CR2->RD_WRN`, and based on the RM0490 Rev5, it's indicated the Transfer direction (controller mode). It is normal for the bit to be 1 when the I2C_GENERATE_START_READ request is configured.

This may be a particular case for this sensor, could you please share the Infineon user manual (I couldn't find it), in order to share it
with our team, as well as a snippet of code for further analysis.

With regards,

AndrewWern, 2025-09-03:
> Hello [@AndrewWern](https://github.com/AndrewWern),
>
> Thank you for the report.
>
> Regarding the issue, and after a first analysis, the API HAL_I2C_Mem_Read_IT() is used to read data from a specific [...]

It can be found here [link to product page](https://www.infineon.com/part/TLE493D-W2B6-A1#documents) and is also attached

[infineon-tle493d-w2b6-um-v01-01-usermanual-en.pdf](https://github.com/user-attachments/files/22124108/infineon-tle493d-w2b6-um-v01-01-usermanual-en.pdf)